rm-hull/luma.led_matrix #276: Segmentation Fault running basic examples

SinnGit, 2022-05-22:
This is similar to closed issue #255 , I'm experiencing the same error when running a basic example.  The panel lights up then immediately goes dark returning the "Segmentation Fault" error in the terminal when running
 `sudo python3 filename.py`

I've formatted the SD Card and used a fresh install of Raspian 32 bit full OS and experienced the same behavior.

I've attempted to update the WS2812 package and am receiving errors when doing so.

ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-djflmpk8/ws2812_bb4b8700532f4992a0be6183142b1eea/setup.py'"'"'; __file__='"'"'/tmp/pip-install-djflmpk8/ws2812_bb4b8700532f4992a0be6183142b1eea/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-lvd2lur5/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.9/ws2812 Check the logs for full command output.

#### Python Version
Python 3.9.2

#### Type of Raspberry Pi
RPI4

#### Linux Kernel version
Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux

#### Expected behaviour
LED Matrix Lights as coded

#### Actual behaviour
LED Matrix lights up and immediately goes dark, terminal returns "Segmentation fault" when the code stops executing.  I experience the same behavior whether I define a matrix or LED strip.  Worth noting I have no issues when leveraging the adafruit neopixel libraries, the panel behaves as expected.    

Here is the basic example I am using.
`from luma.core.render import canvas`
`from luma.led_matrix.device import neopixel`
`device = neopixel(cascaded=512)`
`with canvas(device) as draw:`
`    draw.point((0,0), fill="white")`

Thanks in advance

thijstriemstra, 2022-05-23:
can you try a different version of python 3?

thijstriemstra, 2022-05-23:
>  I’m happy to upgrade the python version as a test, but this does not seem like a legitimate fix, 

Sounds good, I think it might help fix the issue because it has before.

Father001, 2022-05-24:
I get the same error, my python version is 3.7,  maybe need to close Audio output ,  you can read this:  https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/
when I closed Audio output, it worked well.
I hope it can help you.

SinnGit, 2022-05-24:
Thanks all.  I've tried both suggestions, disabling audio output and upgrading to python 3.10.4.

After disabling the audio output there is no change in behavior.  

After upgrading to python and attempting to execute the script, it returns "ModuleNotFoundError: No module named 'luma'".

I attempted to reinstall the luma.led_matrix library and my system is choked attempting to install dependencies.

Collecting rpi-ws281x
  Downloading rpi_ws281x-4.3.4.tar.gz (64 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 64.2/64.2 KB 1.1 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      ERROR: Can not execute `setup.py` since setuptools is not available in the build environment.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

I ran `sudo python3 -m pip install --upgrade setuptools` to reinstall the setup tools and was able to get past the errors installing the luma.led_matrix/rpi-ws281x library.  Eventually I receive the following errors while the cbor library is being installed

      ModuleNotFoundError: No module named '_ctypes'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

I'm stuck attempting to install _ctypes, which should be included as part of Python3 from my reading.  I think it is safe to say attempting to upgrade python on raspberry pi OS caused a cascade of issues.  

I'm formatting the SD card and starting over, I was able to install the module with a fresh install of Raspberry Pi OS.