rm-hull/luma.led_matrix #277: Possible to Increase the Font Size?

ASparkOfFire, 2022-06-08:
Is it Possible to Increase the Font size?

SN-8023, 2022-06-08:
Just create your own character set...

ASparkOfFire, 2022-06-29:
> Just create your own character set...

Can you please guide me?

ASparkOfFire, 2022-06-30:
Okay so I managed to create custom character set but the problem is they are cut in half. I want 16x16 Characters for 64x16 matrix. @rm-hull Sorry for bothering but is there any way to use 16x16 characters?

ChrisToxz, 2022-07-17:
I have no idea, but I'm using a custom font through pillow.
Currently I'm using [Silkscreen](https://www.dafont.com/silkscreen.font) and [White Rabbit](https://www.dafont.com/white-rabbit.font) for the bigger text

```
from PIL import *
font = ImageFont.truetype("slkscr.ttf", 8) # Where 8 is the font size
with canvas(device) as draw:
    draw.text((0, 0), "Text", fill="white", font=font)
```