rm-hull/luma.led_matrix #282: Is there a limit on cascaded LED's?

Letsamsiyu, 2022-10-13:
Hi,
first of all thanks for the great library. I do have a question about limitation because I did not find any info on that.
I want to build a fairly long display with the [16x16 neopixel led matrix](https://www.adafruit.com/product/2547) (to be precise: 36 of them in series). So the total width is 36*16=576 px in width and 16 in height, a total of 9216 LEDs.

The limit seems to be 2728. I wrote a simple crawl example and if I change the 2728 to anything higher the first module makes a quick flash and then nothing. Is there some kind of limitation?

#### Test crawl script
```
import time

from luma.led_matrix.device import neopixel
from luma.core.render import canvas

cascaded_leds = 2728        #works
#cascaded_leds = 9216      #does not work

device = neopixel(cascaded=cascaded_leds)
j=0
while True:
    for i in range(device.cascaded):
        with canvas(device) as draw:
            draw.point((i, 0), fill="green")
            print(j,i)
    print(j)
    j=j+1
```

#### Type of Raspberry Pi
Hardware	: BCM2711
Revision	        : b03111
Serial		: 100000000f1365e9
Model		: Raspberry Pi 4 Model B Rev 1.1
RAM                : 2 GB

#### Linux Kernel version
Linux corpi0000 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

#### Expected behaviour
If I put 9216 as cascaded it should crawl to the end.

#### Actual behaviour
Any value higher than 2728 does not work.

rm-hull, 2022-10-13:
Nothing specific in the library as far as i know. 
How are you powering the LEDs? Presumably not with the 5v supply from the RPi?

Letsamsiyu, 2022-10-14:
So, after some research we found that a pixel needs a 24 bit package. So if there are 2728 pixels its 65472 bit, 64 short for 2 ^ 16 which is 65536. The 64 might be some overhead?  
So we assume that there might be a unsigned short int somewhere in an underlying C lib? Does this make any sense?

rm-hull, 2022-10-14:
Sounds plausible ... might be worth checking with the C lib authors: https://github.com/jgarff/rpi_ws281x

Letsamsiyu, 2022-10-15:
Ah, thank you very much for the link, I will check!

Letsamsiyu, 2022-10-17:
So it seems that there is a limit on cascaded leds https://github.com/jgarff/rpi_ws281x/pull/135#issuecomment-263017900

> Both PWM and PCM use DMA transfers (means no CPU involvement for the transfer).
Max DMA transfer length is 65536 bytes. For PWM this has to cover 2 led strands, for PCM only one. With 12 bytes per LED (4 colors, 8 symbols per color, 3 bits per symbol) this means approximately 5400 LEDs for a single strand in PCM and 2700 LEDs per string for PWM.
I suppose you'll never reach the max due to the power needed and signal degradation.
SPI uses the SPI device driver in the kernel. For transfers larger than 96 bytes the kernel driver also uses DMA.
So I expect all three variants to be very similar in CPU load.

closed 2022-10-17 by Letsamsiyu