rm-hull/luma.led_matrix #288: Is luma.led_matrix compatible with Raspberry Pi 64-bit OS?

linuxminty, 2023-04-13:
#### Type of Raspberry Pi
4B

#### Linux Kernel version
Linux raspberrypi-4B 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

#### Expected behaviour
Hoping the 7-segment display and luma.led_matrix code will work with the Raspberry Pi 64-bit OS.

#### Actual behaviour
When I run the code for luma.led_matrix, the following error is displayed:
```
Traceback (most recent call last):
  File "/home/piguy/luma7datetime.py", line 15, in <module>
    from luma.led_matrix.device import max7219
ModuleNotFoundError: No module named 'luma'

```
Thank you for your help.

thijstriemstra, 2023-04-13:
yes it is. how did you install luma.led_matrix?

linuxminty, 2023-04-14:
After posting this issue, I reinstalled luma.led following the instructions from `https://github.com/rm-hull/luma.led_matrix/blob/master/doc/install.rst`
I'm happy to report that the device now appears to be working as it should with Raspberry Pi 64-bit OS. This issue can be closed.
Thank you for taking the time to reply, thijstriemstra. I should have tried to reinstall luma.led_matrix before posting. Sorry!

closed 2023-04-14 by rm-hull