rm-hull/luma.led_matrix #293: Error when trying to install Luma Led Matrix with Raspberry Pi OS Bookworm

linuxminty, 2024-01-12:
I'm trying to get the seven-segment display to work with my RPi 4, running Bookworm. (It worked fine with Bullseye.) 

#### Type of Raspberry Pi
Model 4, 1GB

#### Linux Kernel version
Linux raspberrypi-4B 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

#### Actual behaviour
Using the instructions to install from PyPi,

`sudo -H pip install --upgrade --ignore-installed pip setuptools` shows this
```
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```
Please provide guidance on how to get the Luma Led Matrix properly installed on this Pi. Thank you.

thijstriemstra, 2024-01-12:
Follow these instructions: https://github.com/rm-hull/luma.oled/blob/master/doc/software.rst#software but install luma.led_matrix instead of luma.oled in last step.

linuxminty, 2024-01-12:
Thank you for your prompt reply and suggestion. I hope I'm following your directions correctly and in the right order: (1) created the virtual environment, (2) installed the latest version of the library, then (3) attempted to install the dependencies, which ended with the error message below.
```
sudo apt-get install python3 python3-pip python3-pil libjpeg-dev zlib1g-dev libfreetype6-dev liblcms2-dev libopenjp2-7 libtiff5 -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libtiff5

```

linuxminty, 2024-01-16:
As you suggested, I'm trying to follow the instructions in the link you provided, but am unsuccessful. I'm unsure about what you refer to as the "last step," where I'm to "install luma.led_matrix instead of luma.oled."

Here is the output of the steps I've tried:
First, I ran `$ python3 -m venv ~/luma-env`
No output was shown, but a virtual environment was created in the home directory as expected.

Second, I ran 
`$ ~/luma-env/bin/python -m pip install --upgrade luma.led_matrix` 
(with luma.led_matrix substituted for luma.oled)

The output is:
```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: luma.led_matrix in ./luma-env/lib/python3.11/site-packages (1.7.1)
Requirement already satisfied: luma.core>=2.4.0 in ./luma-env/lib/python3.11/site-packages (from luma.led_matrix) (2.4.1)
Requirement already satisfied: pillow>=9.2.0 in ./luma-env/lib/python3.11/site-packages (from luma.core>=2.4.0->luma.led_matrix) (10.2.0)
Requirement already satisfied: smbus2 in ./luma-env/lib/python3.11/site-packages (from luma.core>=2.4.0->luma.led_matrix) (0.4.3)
Requirement already satisfied: pyftdi in ./luma-env/lib/python3.11/site-packages (from luma.core>=2.4.0->luma.led_matrix) (0.55.0)
Requirement already satisfied: cbor2 in ./luma-env/lib/python3.11/site-packages (from luma.core>=2.4.0->luma.led_matrix) (5.5.1)
Requirement already satisfied: RPI.GPIO in ./luma-env/lib/python3.11/site-packages (from luma.core>=2.4.0->luma.led_matrix) (0.7.1)
Requirement already satisfied: spidev in ./luma-env/lib/python3.11/site-packages (from luma.core>=2.4.0->luma.led_matrix) (3.6)
Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in ./luma-env/lib/python3.11/site-packages (from pyftdi->luma.core>=2.4.0->luma.led_matrix) (1.2.1)
Requirement already satisfied: pyserial>=3.0 in ./luma-env/lib/python3.11/site-packages (from pyftdi->luma.core>=2.4.0->luma.led_matrix) (3.5)
```
If these are the correct steps/commands, what is the next step? Do I now need to go to the "Installing from PyPi" instructions? Thank you for any assistance you can offer.

thijstriemstra, 2024-01-16:
you need to use `~/luma-env/bin/python sevensegment_demo.py`

linuxminty, 2024-01-16:
running `~/luma-env/bin/python sevensegment_demo.py`
results in:
`/home/mrpi/luma-env/bin/python: can't open file '/home/mrpi/sevensegment_demo.py': [Errno 2] No such file or directory
`

linuxminty, 2024-01-17:
I'm still trying to get the ZeroSeg display to work wth Raspberry Pi OS Bookworm.

I cloned the luma.examples from https://github.com/rm-hull/luma.examples and then tried to run sevensegment_demo.py with the result:

```~/luma-env/luma.examples/examples $ python3 sevensegment_demo.py
Traceback (most recent call last):
  File "/home/piguy/luma-env/luma.examples/examples/sevensegment_demo.py", line 14, in <module>
    from demo_opts import get_device
  File "/home/piguy/luma-env/luma.examples/examples/demo_opts.py", line 8, in <module>
    from luma.core import cmdline, error
ModuleNotFoundError: No module named 'luma'
```
Earlier, I tried to reinstall from PyPi https://luma-led-matrix.readthedocs.io/en/latest/install.html#installing-from-pypi
`$ sudo apt install build-essential python3-dev python3-pip libfreetype6-dev libjpeg-dev libopenjp2-7 libtiff5 `resulted in:
```
sudo apt install build-essential python3-dev python3-pip libfreetype6-dev libjpeg-dev libopenjp2-7 libtiff5
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libtiff5

```

I don't know what else to try. Help, please.

linuxminty, 2024-06-09:
Now, several months after my initial efforts, this issue remains unresolved for me and I continue to try to get luma.led to work with my ZeroSeg 7-segment display and Raspberry Pi 4 running Bookworm. To that end, here are the recent steps I've taken, listed in order:
1. Reinstalled Bookworm Raspberry Pi OS.
2. Ran `python3 -m venv ~/luma-env `
3. Ran `~/luma-env/bin/python -m pip install --upgrade luma.led_matrix`
4. Cloned the luma.examples from https://github.com/rm-hull/luma.examples to `~/luma-env/bin`
5. Ran `~/luma-env/bin/luma.examples/examples $ python3 sevensegment_demo.py`
with the following result:

```
Traceback (most recent call last):
  File "/home/piguy/luma-env/bin/luma.examples/examples/sevensegment_demo.py", line 14, in <module>
    from demo_opts import get_device
  File "/home/piguy/luma-env/bin/luma.examples/examples/demo_opts.py", line 8, in <module>
    from luma.core import cmdline, error
ModuleNotFoundError: No module named 'luma'
```
I'm missing something, but don't know what it is. What do I need to do to run luma.led successfully with Bookworm?

 Thank you for any guidance.

SeagullPz, 2024-06-17:
https://luma-led-matrix.readthedocs.io/en/latest/install.html#pre-requisites

**NO sudo anywhere** 

pip install --upgrade --ignore-installed pip setuptools --break-system-packages

python3 -m pip install --upgrade luma.led_matrix --break-system-packages

#Ran my 8x8 matrix example via “python3 small2a.py”

#Warning installed in '/home/david/.local/bin' which is not on PATH
#So I added it to PATH

I’m a total Linux noob.

This is surely the wrong solution ( --break-system-packages ) !!!!! but it worked for me.

linuxminty, 2024-10-26:
Why is this marked as "Completed"? It is still unresolved for me, as noted in my last post of June 9. 

I didn't follow the comments by SeagullPz, as they didn't appear to be specific for the seven-segment display, which I'm trying to use with Raspberry Pi OS Bookworm.

Specific details of what to do (steps to follow) for the the seven-segment display to run under Bookworm would be welcome. Thank you again.

thijstriemstra, 2024-10-26:
> This is surely the wrong solution ( --break-system-packages ) !!!!! but it worked for me.

I guess I read the wrong comment.