rm-hull/luma.led_matrix #297: MAX7219 snake pattern

ChrisToxz, 2025-05-21:
I have 24 8x8 LED modules in a 8x3 setup (8 columns, 3 rows)
**Is it possible to drive multiple MAX7219 in a snake pattern?**

I see apa102 & ws2812 supports a mapping parameter, but MAX7219 does not.

I would like to have my modules connected like:
Row 0 → [0][1][2][3][4][5][6][7]
Row 1 ← [15][14][13][12][11][10][9][8]
Row 2 → [16][17][18][19][20][21][22][23]

I'm working on a custom PCB board and with above layout, the connection between module 7 and 8 can be as short as possible, rather than going from one side to the other side of the PCB.

Am I missing something in the docs and is this already possible? If not, any headups how I could achieve this anyhow?

rm-hull, 2025-05-23:
from memory, the apa mapping is per single pixel so relatively straightforward.

form max7219, it would need to be per 8x8 pixel block - so the existing wouldnt work, but it should be pretty easy to do - it might be as simple as providing a different set of offsets, here:

https://github.com/rm-hull/luma.led_matrix/blob/main/luma/led_matrix/device.py#L80-L82

ChrisToxz, 2025-05-23:
Thanks for the head ups.
I have been tinkering with that for the past few days, but with some help from AI I got the following solution which flips and turns around the 'image' for even rows.

Did some benchmarking and the losses are acceptable. 160FPS vs 175 FPS.

```py
class SnakedMax7219(max7219):
    def display(self, image):
        w, h = image.size
        band_h = 8
        img = image.copy()

        for row in range(h // band_h):
            if row % 2 == 1:  # odd rows
                y0 = row * band_h
                y1 = y0 + band_h
                band = image.crop((0, y0, w, y1))
                band = band.transpose(Image.ROTATE_180)
                img.paste(band, (0, y0))

        super().display(img)

width = 64
height = 24
device = SnakedMax7219(
    serial,
    width=width,
    height=height,
    block_orientation=-90,
    contrast=1,
)   
```

Found also a way that overwrites the offset as you suggested, with no performance losses, though it would require to 
physically rotate my modules 180 degrees, for the even rows. Didn't got that working software wise.
E.G on the module Row 1, Col 1 was lit up, while it should be actually Row8,Col8.

In case you have a suggestion to get that still working with the offset overwrite, I'm all ears.
I'm not so into such low level coding!

The offset manipulation:
```py
class SnakeMax7219(max7219):
    def __init__(self, serial_interface, width=64, height=24,
                 rotate=0, block_orientation=0, contrast=0x70, **kwargs):
        super().__init__(serial_interface,
                         width=width, height=height,
                         rotate=rotate,
                         block_orientation=block_orientation,
                         contrast=contrast,
                         **kwargs)

        w, h = self._w, self._h
        new_offsets = []
        rows = list(range(h - 8, -8, -8))     
        for row_idx, y in enumerate(rows):
            xs = list(range(w - 8, -8, -8))   
            if row_idx % 2 == 1:
                xs.reverse()                   
            for x in xs:
                new_offsets.append((y * w) + x)

        assert len(new_offsets) == self.cascaded
        self._offsets = new_offsets
```